Spark-Concepts/xPRO #6: Limit switch current

muxa, 2016-09-30:
I'm wondering what current will be going through limit switches, so i can use appropriate diameter wire.

Arctic-Stingray, 2016-09-30:
I would not go smaller than 20awg

Arctic-Stingray, 2016-09-30:
I believe that the limit switch circuit on the xpro is 5 volts DC but I
would need to check it to be sure. I was running into issues with false
triggers due to wire size and length of run of the cables so I used solid
state relays in my control cabinet.  Doing this allowed me to keep the
cable distance from the xpro to the relay down-to less than 6 inches. I
just drive the solid state relays with my power supply and run the 24 volts
DC through the limit switchs.

muxa, 2016-09-30:
Thanks @Arctic-Stingray 
For others in metric system: 20 AWG is 0.81mm diameter

muxa, 2016-10-12:
Reopening in case somebody else is interested in this (it's harder to find closed issues)

muxa, 2016-11-11:
I'm getting spontaneous hard limit signals while running the machine. I've been advised to use shielded cable.