Spark-Concepts/xPRO #7: Connecting a laser

muxa, 2016-10-12:
I'm thinking of connecting a laser to my xPro, but not sure how to wire it and how to control it from G-Code. The laser that i'm looking at is https://www.aliexpress.com/item/10W-laser-head/32696422108.html?spm=2114.13010308.0.0.ifkm09
Am i correct in thinking that the D11 pin (Variable Spindle PWM) should be used to control the intensity of the laser?  
Any help is appreciated.

NZ-Mark, 2017-03-24:
Hi Muxa,

I did exactly that. I had previously purchased a cheap A4 sized CNC laser kit that had a 2.5W laser.
I later bought a Ooznest CNC router kit 750mm x 1m which uses the Spark Concepts xPro board. I made an adapter to fit the laser to the Z axis instead of the router.

To control the laser, I used the Spindle out from the board to drive a power FET that turned on / off the laser.

I use Fusion 360 (autocad) to do the CAD and CAM and I modified a grbl post processor to work with the laser. Basically I removed all Z axis movement and added code to control the spindle - on / off whenever a line / arc etc. was generated in Gcode.

When running the CAM part of Fusion, I created a tool - being a milling bit that had a very small diameter e.g. 0.1mm, then did the standard CAM and GCode generation using the post processor I created.

It seems to work well.

Here's an example of it at work. I wanted to mark a grid on the base board of the router table, so I used the laser. [image]

Mark

muxa, 2017-03-25:
Thanks, @NZ-Mark, i'll give this a try. 
By the way, did you know that Fusion 360 has experimental support for laser cutters? You'll need to enable it in Preferences > CAM

NZ-Mark, 2017-03-25:
Hi Muxa,
I'll check that out in Fusion.

Incidentally, for the grid (5cm x 5cm square) that I 'burned' into the spolier board as shown in the photo above: I wrote the G-code from scratch, or rather I wrote a program that wrote the G-Code.  The code is attached. It's heavily commented and you can see how I turn on and off the Laser by setting the spindle speed. The feed speed is set so that the line was well defined, but didn't excessively burn the surface. 

I have used the laser to cut out designs I've created in Fusion - cutting the design out of paper using the laser as a test before attaching the router to do the real cut. I still have a bit more experimentation to do with the laser to get the speeds etc right so that it cuts Ok without setting the paper on fire! I find the 2.5W laser is good for marking wood etc, but is way under powered for cutting.

Mark

[Grid_for_LASER.txt](https://github.com/Spark-Concepts/xPRO/files/869873/Grid_for_LASER.txt)

Mikelawler, 2023-05-28:
Hi, I've had an ooznest ox for the past 4 years and have just decided to look into adding a laser to my cnc. Searching through Google and I've stumbled onto this conversation. I wonder if you would mind helping me out with it ?

I'm not sure which laser is what I want yet, it's the xpro connections that have me baffled. Could you please explain the connections on the board and how I would use them to switch the laser on/off. Any help you could offer would be great.

Thanks again,

Mike.

NZ-Mark, 2023-05-29:
Sorry Mike, I can't be of much help as I sold my ooznest ox a few years back. The other thing was that the ooznest circuit board failed soon after I got that the machine. I wasn't impressed by this. The onboard power regulator failed and everything on the board was fried. I ended up getting a grbl board and stepper driver boards from Ali Express, rewired and upgraded all the electronics. I also put everything in a better dust proof box under the machine. It was when I was reconfiguring the electronics that I set up the laser side of things.

But I think you're on the right track. I think I did use the spindle speed PWM to control the laser and used GRBL codes to set the speed to either 0 or 100%

NZ-Mark, 2023-05-29:
And a bit more on that. If you look at the Grid_for_Laser.txt file in a previous post, you can see the Laser on/off control

S0 (LASER Off)
G0 (Rapid position)
X200 Y00 Z0 F750 (Position for next line)
G1 (Linear interpolation)
S2048 (LASER On)

Mikelawler, 2023-05-30:
Thanks for the reply Mark. I'm actually quite surprised to hear about your problems with the ox. I've been running mine without any problems since I bought it. I agree with changing the dust proof box though as the one I received wasn't very good so I printed myself another.

Thanks for the information, I'll definitely look into the pwm connection once I decide on the laser module. I'm hoping it'll be a better substitute for the k40 that I've just sold.

Kind regards 

Mike.