Spark-Concepts/xPRO #9: Grbl v1.1

muxa, 2016-12-13:
Is xPro v2 compatible with Grbl v1.1?

woale, 2017-03-17:
@muxa, meanwhile did you already gave grbl v1.1 a try? I'm also tempted to see how "Dynamic Laser Power Scaling with Speed" works ...

muxa, 2017-03-20:
@woale no, i haven't. I'm worried that i won't be able to restore it back to an earlier version if something goes wrong. 
I'm very interested in the new real-time jogging mode.

ndevenish, 2017-03-27:
We've been running the xPro 2 with GRBL 1.1 and haven't had any issues with the control board - haven't tried downgrading but doubt there would be any issue as GRBL doesn't overwrite the bootloader (the part responsible for reflashing).

Mainly the issues have been finding software that supports it - e.g. Chillipeppr doesn't work, bCNC doesn't use the new Jogging (a pain when jogging into the soft limits and needing to restart&rehome). UGS works nicely at least.

muxa, 2017-03-28:
I've done the upgrade today and it went flawlessly (just had to restore the settings after upgrading). Thanks, @ndevenish

closed 2017-03-28 by muxa