Spark-Concepts/xPRO #14: GRBL MPOS RANDOMLY CHANGED ???

blayno, 2017-10-27:
Hey peeps
ive had my ox cnc for a while now, it has limit/homing switches installed, im using bcbc as my commander program, untill 2 days ago everything has been working perfect, but now when i home my machine, my X axis MPOS coords are showing completely different values than 2 days ago, which i driving me nuts as its stuffed up my settings in bcnc. , from my understanding, you cant even program your MPOS coordinates so why the hell has it just randomly changed. previously it was telling me my X axis was at -589.989 after homing, but now it tells me its at -55 which now shows the x axis homed at the right hand side of the
bcnc work area screen, when before it was the left, i am totally confused what has happened, can anybody out there help me please ?? im about to throw my cnc in the bin !!!!

Spark-Concepts, 2017-11-02:
So it sounds like there is an issue with noise on the usb line and possibly some corruption of the eeprom.  My first suggestion would be to check settings and make sure they are what you expected them to be.   If they are different,  reset the board with "RST=*" command.   Reapply settings,  and check for sources of noise near the usb cable - near switching psu, near router ac cable,  near motor lines, etc). 

If they are fine, try running the board grin universal gcode sender to rule out anything with bcnc.  

Lastly,  try reflashing the board.  

Feel free to email me at Mike@spark-concepts.com if you need clarification on any step. 

Thanks,
Mike

blayno, 2017-11-02:
ok the problem is not noise, i have tried many programs, ugs versions, white rabbit, bcnc etc, i have my xpro v3 inside with me, trying to connect to ugs again, ugs wont connect, plug xpro in to computer and get usb verification sound, try to connect with ugs and get this **** Connected to COM3 @ 115200 baud ****, then if i try and enter $$ command it tells me grbl hasnt finished booting, i cant do anything with the xpro v3 cheers, i have pressed reset, and still nothing
im pretty sure this board is totaled buddy

Feel free to email me at Mike@spark-concepts.com<mailto:Mike@spark-concepts.com> if you need clarification on any step.

blayno, 2017-11-02:
ive had this board over 6 months, and from day 1 ive had nothing but problems with it, im pretty sure i got a lemon bud ! its had a mind of its own the whole time, ive ordered a cheap arduino and cnc shield as i have orders i need to get out to people

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:43 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ok the problem is not noise, i have tried many programs, ugs versions, white rabbit, bcnc etc, i have my xpro v3 inside with me, trying to connect to ugs again, ugs wont connect, plug xpro in to computer and get usb verification sound, try to connect with ugs and get this **** Connected to COM3 @ 115200 baud ****, then if i try and enter $$ command it tells me grbl hasnt finished booting, i cant do anything with the xpro v3 cheers, i have pressed reset, and still nothing
im pretty sure this board is totaled buddy

blayno, 2017-11-02:
it also is smelling hot now too, only when being plugged into the pc

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:45 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ive had this board over 6 months, and from day 1 ive had nothing but problems with it, im pretty sure i got a lemon bud ! its had a mind of its own the whole time, ive ordered a cheap arduino and cnc shield as i have orders i need to get out to people

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:43 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ok the problem is not noise, i have tried many programs, ugs versions, white rabbit, bcnc etc, i have my xpro v3 inside with me, trying to connect to ugs again, ugs wont connect, plug xpro in to computer and get usb verification sound, try to connect with ugs and get this **** Connected to COM3 @ 115200 baud ****, then if i try and enter $$ command it tells me grbl hasnt finished booting, i cant do anything with the xpro v3 cheers, i have pressed reset, and still nothing
im pretty sure this board is totaled buddy

blayno, 2017-11-03:
________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:49 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

it also is smelling hot now too, only when being plugged into the pc

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:45 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ive had this board over 6 months, and from day 1 ive had nothing but problems with it, im pretty sure i got a lemon bud ! its had a mind of its own the whole time, ive ordered a cheap arduino and cnc shield as i have orders i need to get out to people

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:43 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ok the problem is not noise, i have tried many programs, ugs versions, white rabbit, bcnc etc, i have my xpro v3 inside with me, trying to connect to ugs again, ugs wont connect, plug xpro in to computer and get usb verification sound, try to connect with ugs and get this **** Connected to COM3 @ 115200 baud ****, then if i try and enter $$ command it tells me grbl hasnt finished booting, i cant do anything with the xpro v3 cheers, i have pressed reset, and still nothing
im pretty sure this board is totaled buddy

blayno, 2017-11-03:
thats where its getting hot, right in the middle of the pic, but its hot on the underneath of the board when plugged into usb port, i have tried several ports, even another pc, and also changed cables

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:49 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

it also is smelling hot now too, only when being plugged into the pc

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:45 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ive had this board over 6 months, and from day 1 ive had nothing but problems with it, im pretty sure i got a lemon bud ! its had a mind of its own the whole time, ive ordered a cheap arduino and cnc shield as i have orders i need to get out to people

________________________________
From: blayne hixson <blaynehixson@hotmail.com>
Sent: Friday, 3 November 2017 10:43 AM
To: Spark; Spark-Concepts/xPRO; Spark-Concepts/xPRO
Cc: Author
Subject: Re: [Spark-Concepts/xPRO] GRBL MPOS RANDOMLY CHANGED ??? (#14)

ok the problem is not noise, i have tried many programs, ugs versions, white rabbit, bcnc etc, i have my xpro v3 inside with me, trying to connect to ugs again, ugs wont connect, plug xpro in to computer and get usb verification sound, try to connect with ugs and get this **** Connected to COM3 @ 115200 baud ****, then if i try and enter $$ command it tells me grbl hasnt finished booting, i cant do anything with the xpro v3 cheers, i have pressed reset, and still nothing
im pretty sure this board is totaled buddy

Spark-Concepts, 2017-11-03:
So just to clarify,  was everything working perfect up until 3-4 days ago or has it been having issues for 6 months?  The mind of its own really does point itself to an issue with noise on the data lines.  The upset in home position is likely due to a change in the max travel settings or some eeprom values.  

 If you would like to send the board in,  the techs can investigate what is happening and see about repair.   To do this please submit a contact form here - http://www.spark-concepts.com/contact-us-1/

blayno, 2017-11-03:
not perfectly, but yes it was working, now its not, sometimes would work fine, and other times would have a mind of its own for example,my cnc has destroyed alot of my pieces some being 8 -10 hrs in length, randomly just drilling a hole straight through the material and then wasteboard during a pocketing op, or for instance cutting along the Y axis then z not retracting and it tries to cut through like 10mm of material but cant and spins the endmill inside the collet untill it rips it out , i have been through many controller programs, UGS Builds, BCNC, white rabbit etc , i have changed computers twice as i thought maybe they werent powerful enough to run either the cnc or the programs, but nothing ive done has made any difference. i never changed any max travel values on my board or bcnc,  i did fall asleep with xpro v3 connected to the pc  the day before the homing went out of whack, im currently not working and have $0 to my name so i cannot afford to send it off to you or pay for repairs, hence the reason i bought a $25 arduino
thank you

CriticalPoint, 2019-01-24:
> randomly just drilling a hole straight through the material and then wasteboard during a pocketing op

I had this on mine, driving it with the xPRO v3. I sorted this by adding a USB cable that provides power and clears noise. It's a cheap test if nothing else ¯\\_(ツ)_/¯