Spark-Concepts/xPRO #16: My xPRO v3 appears to have fried

ENGR-mike, 2018-01-08:
I have been using a CNC xPRO V3 board successfully with my OpenBuilds MiniMill for the past several months without any problems. I recently hooked up an accessory board (made by a friend) containing MOSFETS to control my spindle (i.e. router) and solenoid valve for coolant. I tried hooking up the spindle signals first. The xPRO worked fine with the accessory board and a relay to turn the spindle on and off. I used the exact same parts (MOSFETS, relay, diodes, etc.) and configuration shown on the Github site. I proceeded to hookup the solenoid (yes I used a flyback diode). When turning the power back on to the board after connecting the solenoid through the accessory board, I think I heard a pop and smelled something burning. I noticed I no longer had lights on for the stepper drivers, but the 24V, 5V and USB comm lights were on. I could still turn the spindle on and off, and I could now activate and deactivate the solenoid. However, I still had no lights on for the stepper drivers and I could not move my stepper motors at all. I tried everything I could think of by disconnecting and reconnecting things. I checked voltages and all are correct (when activating either the spindle or solenoid, I get 5V from the xPRO to my accessory board for the spindle signal and the coolant signal and 24V out of my accessory board for each). I can still turn the spindle on and off and activate/deactivate the solenoid, but no power to the stepper drivers. Did I just fry my xPRO? If everything else is working, what could have happened? Can anyone suggest some diagnostic techniques to help me? Please keep in mind I'm not very electrically inclined.

Spark-Concepts, 2018-01-08:
First thing would be to measure the stepper motor 24V power directly:

1. Set your multi-meter to Volts
2. Look at the Backside of the xPro (disconnect USB cable and Accessory board)
3. Place the Red lead on the unpopulated pad  labelled "12-24V (OUT)" 
4. Place the Black lead on the unpopulated pad  labelled "GND"
5. Briefly turn xPro Power On while testing for 24VDC (*be careful not to short test leads while performing power-on test)... 

With the accessory board, it's important that the switched side of the relay be electrically isolated from the solenoid action: i.e. the Spindle wiring neutral or Gnd should be electrically isolated from the xPro's 24V/5V Ground reference (assuming you have a separate power supply for your spindle):

1. Disconnect your accessory board from the xPro
2. Set multi-meter on Ohms or Continuity (beeps when leads are touched)
3. Check for continuity on accessory board Ground to each lead of the relay switch to the spindle wiring
4. If there's continuity (beeps or 1-2 ohms) then this is potentially a problem...

...You can also try toggling the step enable invert from $4=1 to $4=0 (unlikely)

Without seeing your setup and wiring first hand, it's going to be difficult to diagnose any further.  It's also possible a stray wire may have inadvertently caused a short - I typically suggest either solder tinning or using wire ferrules when connecting wires to the xPro screw terminals (i.e. Similar to this kit: http://a.co/6LMwmF0).

If all else fails, contact: mike@spark-concepts.com - send pictures of your wiring, power supplies, and accessory board spindle configuration.  If it turns out your board is fried, I'd be more than happy to look at it.

ENGR-mike, 2018-01-08:
Let me know if the following helps:
[image]

ENGR-mike, 2018-01-08:
P.S.  I do solder tin all my leads inserted into the screw terminals.

Spark-Concepts, 2018-01-09:
Nice Schematic! You may be aware, however the one thing that draws my attention is the RESET/ABORT e-stop switch... your schematic shows a 'Normally Closed' switch... it should, and may already be 'Normally Open" when not depressed... pulling the RESET/ABORT down to Ground will disable the steppers.   Also the banded ends of the flyback diodes should be toward Vcc (24VDC).  According to your schematic it appears the flyback diode on the 24VDC solenoid valve my be wired backwards (this could potentially damaged the associated MOSFET and/or diode)... Essentially this would short out your 24VDC when actuated.

ENGR-mike, 2018-01-09:
Thanks!  I did notice that the flyback diode for the coolant solenoid was backwards, but it is wired correctly.  Also, I did notice that a NC switch for the RESET/ABORT was not correct and I did install a NO switch.  I just forgot to update the above schematic.
With all of that said, I don't have a fried board.  I found a broken/shorted wire on a DIN connector for the z-axis leads that cause the board to act like I just pressed the RESET/ABORT.  I fixed the wire and pressed the RESET/ABORT then reset it and all was well.  Everything works as it should again, including the coolant solenoid and the spindle on/off signals.
Thanks for the help and support.  I also appreciate the patience with this gear head.

closed 2018-01-09 by ENGR-mike